cesanta/mongoose-os #617: mos flash and wifi config return read err EOF

itdog1973, 2024-11-19:
Hi, I am not sure why the command that I run on mos with my esp32 keep returning read err 

[image]

could anyone help?

scaprile, 2024-11-19:
Please don't open an issue to ask a question.
Please no useless snapshots
How do you think anyone can guess what you are doing wrong when you don't show what you are doing ?
There are docs 
There is a forum https://community.mongoose-os.com/latest
There is a chat https://app.gitter.im/#/room/#cesanta_mongoose-os:gitter.im

itdog1973, 2024-11-19:
ok. sorry for the misuse of the issue 

> Please don't open an issue to ask a question. Please no useless snapshots How do you think anyone can guess what you are doing wrong when you don't show what you are doing ? There are docs There is [...]